LORD-MicroStrain/MSCL #263: chmod: cannot access 'install-cp210x-0.2.sh': No such file or directory

saytesla, 2022-06-29:
Hi,
I'm trying to install the linux drivers. I am following this link (https://github.com/LORD-MicroStrain/Drivers/releases/tag/linux-cp210x-0.2)

I am getting this error "chmod: cannot access 'install-cp210x-0.2.sh': No such file or directory"
when I run command : chmod a+x install-cp210x-0.2.sh

msclissa, 2022-06-29:
Are you running the command from the same directory where the downloaded [install-cp210x-0.2.sh](https://github.com/LORD-MicroStrain/Drivers/releases/download/linux-cp210x-0.2/install-cp210x-0.2.sh) file is located?

saytesla, 2022-06-29:
Yes so I ran it in the same directory where the file is located. It worked.
But now when I try running sudo ./install-cp210x-0.2.sh i am getting an error.

"Error! Your kernel headers for kernel 5.15.32-v8+ cannot be found.
Please install the linux-headers-5.15.32-v8+ package,
or use the --kernelsourcedir option to tell DKMS where it's located"

msclissa, 2022-06-30:
Can you confirm that the kernel headers are installed on your system? If that doesn't work can you let us know the kernel version and the install directory?

It looks like this page has some good info/instructions on the topic: https://www.tecmint.com/install-kernel-headers-in-ubuntu-and-debian/
**Note:** the above-linked page is not owned or managed by us, be sure to double check that any recommended commands do what they claim before running them on your system.