LORD-MicroStrain/MSCL #278: Initial GPS Clock Time setting with Python API

BrandonNelsenNGC, 2022-08-04:
I'm having trouble setting the GPS Clock on a 3DM-GX5-25 using the Python API. I've been able to create a Timestamp object and use `setTimeNow` to set/sync to the current system time. Now I'm trying to use that with `InertialNode.setGPSTimeUpdate` to set/sync the device time using that so I can get timestamping data from channel sampling packets. Could you provide some example Python implementation for that? I'm getting hung up on specifying the `MipTypes::TimeFrame` argument and whether both time frame options need to be set or simply just seconds.

BrandonNelsenNGC, 2022-08-08:
Worked out the Python implementation and here's how I achieved it:
```
systemTime = mscl.Timestamp.timeNow()
inertialNode.setGPSTimeUpdate(mscl.MipTypes.TIME_FRAME_SECONDS, systemTime.seconds(mscl.Timestamp.GPS))
```
Now I can get the timestamp off the sample packets using the `mipDataPacket.deviceTimestamp()` function.

msclissa, 2022-09-08:
I'm glad you got this figured out - thanks so much for sharing your solution!

The `setGPSTimeUpdate()` function can also be used to set the week number part of the timestamp (`mscl.MipTypes.TIME_FRAME_WEEKS`). The idea behind this is to save bandwidth - if you want to keep everything synchronized by updating the time every second on the second you don't have to waste bandwidth by sending the week number every second.

I'm going to go ahead and close this, but if you have any additional questions feel free to re-open this or open another issue. Thanks!

closed 2022-09-08 by msclissa