LORD-MicroStrain/MSCL #279: Shared GPS Time returns internal system time on 3DM-GQ7

scotts11, 2022-08-29:
I am using the 3DM-GQ7, and I am trying to include the Shared GPS Time (message 0xFF 0xD3) in both the AHRS_IMU data stream and the ESTFILTER data stream).  I am getting GPS Time correctly from the GNSS1 channel (only using one channel right now), but the AHRS_IMU and ESTFILTER data streams are just returning the internal system time (basically seconds since the system was powered on).

One of the channels (ESTFILTER) has the valid bit set for the GPS Shared Time, and the AHRS_IMU channel does not have the valid bit set for GPS Time.  The GNSS1 channel has the valid bit set, and the time looks correct.

Do I need to initial the Shared GPS time?

msclissa, 2022-09-07:
Sorry for the delay!

All you should need to do is tell the device to use the time from GNSS1 for the internal time - this can be done by calling the `setPpsSource()` function on the `InertialNode` object:
```
// set time source to GNSS Rec 1
device.setPpsSource(mscl::InertialTypes::PPS_RECEIVER_1);

// to automatically use GNSS Rec 1 as time source on device startup
device.saveSettingsAsStartup({ mscl::MipTypes::CMD_PPS_SOURCE });
```

I'm surprised that the GPS Time from the filter is being marked valid, I'm not seeing that on my device - I'll bring this up with our firmware team.

Once the PPS source is set it may take a couple seconds for the device to start using it, but from there the values should be marked valid and the time for each data set should be synchronized to the time from GNSS Receiver 1.

Hope this helps, let us know if you run into any issues!

scotts11, 2022-09-08:
That fixed it.  Thanks

scotts11, 2022-09-08:
> External Email Alert
>
> This email has been sent from an account outside of the BAE Systems network.
> Please treat the email with caution, especially if you are requested to click on a link, decrypt/open an attachment, or enable macros.  For further information on how to spot phishing, access “Cybersecurity OneSpace Page” and report phishing by clicking the button “Report Phishing” on the Outlook toolbar.

Thanks, that looks like it did it.  I guess I was just looking in the wrong area of the API.

Scott

closed 2022-09-08 by scotts11