LORD-MicroStrain/MSCL #300: WSDA-2000  Enable Beacon command has failed

ccd50, 2022-10-27:
I am interfacing with WSDA-2000, V-link-200 that has five channels set to sample at 256Hz, using python 2.7 and tynker

Issue: when I am running this code for about 2 - 3 minute I receive the following error. Any help would be appreciative.

(is this issue related to the Beacon or the lost of python thread?)

Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 530, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 483, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:\Python27\10_27_2022.py", line 85, in timeThread
    network.startSampling()
  File "C:\Python27\mscl.py", line 3610, in startSampling
    return _mscl.SyncSamplingNetwork_startSampling(self, *args)
Error_Communication: The Enable Beacon command has failed

## import libraries
import thread
import threading
import Queue
import time
from Tkinter import *
import sys
sys.path.append("../../Python27")
import mscl

root = Tk()

## Set the size of the window
root.geometry("1100x450")

## set label to data on
run_mode=None
def run_mode():
    global run_mode
    label.config(text = None)
    label.config(text=' Data On ')

idle1=None

#put status label on the screen
label = Label(root, text = idle1, bg = "green", bd = 11, fg = "black", font = ("Castellar",14,"bold"))
label.place(x=460, y=130)
label.config(text = None)
label.config(text="idle mode")

#Put Start Button on the screen
start_data = Button(root, text = "Start Data", padx=20, pady=5,fg="Green", font=("Calibri",14,"bold"),command = lambda: [run_mode(),start_th()])
start_data.place(x=150, y=130)

#Put stop button on the screen
button_quit = Button(root, text = "Exit Program", padx=20, pady=5,fg="red", font=("Calibri",14,"bold"),command = lambda: stop())
button_quit.place(x=290, y=130)

# Basestation
basestation = mscl.BaseStation(mscl.Connection.TcpIp("fd7a:cafa:0eb7:6578:0002:51dc::1", 5000))
network = mscl.SyncSamplingNetwork(basestation)
network.addNode(mscl.WirelessNode(20928, basestation))
network.applyConfiguration()
node_a = mscl.WirelessNode(20928, basestation)

# stop/idle function
def stop():
    idleStatus = node_a.setToIdle()
    commQueue.queue.clear()
    root.destroy()

## Communication queue
commQueue = Queue.Queue(maxsize=0)
commQueue.queue.clear()

def timeThread():

    ## setting variables to 0
    ch1_20928data = 0
    ch2_20928data = 0
    ch3_20928data = 0
    ch4_20928data = 0
    ch5_20928data = 0

    while True:
        ## placing data into queue
        commQueue.put(ch1_20928data)
        commQueue.put(ch2_20928data)
        commQueue.put(ch3_20928data)
        commQueue.put(ch4_20928data)
        commQueue.put(ch5_20928data)

        try:
            ## event_generate and append data
            root.event_generate('<<ch1_20928>>', when='tail')
            root.event_generate('<<ch2_20928>>', when='tail')
            root.event_generate('<<ch3_20928>>', when='tail')
            root.event_generate('<<ch4_20928>>', when='tail')
            root.event_generate('<<ch5_20928>>', when='tail')

        except TclError:
            break

        ## syncronize basestation and start sampling @ 500
        network.startSampling()
        sweeps = basestation.getData(500)

        ## perform sweeps and place data into variable for each channel
        for sweep in sweeps:
            data = sweep.data()
            for dataPoint in sweep.data():
                data1 = [sweep.nodeAddress()], [dataPoint.channelName()], [dataPoint.as_string()]
                if data1[0][0] == 20928 and data1[1][0] == 'ch1':
                    ch1_20928data = data1[2][0]

        for sweep in sweeps:
            data = sweep.data()
            for dataPoint in sweep.data():
                data1 = [sweep.nodeAddress()], [dataPoint.channelName()], [dataPoint.as_string()]
                if data1[0][0] == 20928 and data1[1][0] == 'ch2':
                    ch2_20928data = data1[2][0]

        for sweep in sweeps:
            data = sweep.data()
            for dataPoint in sweep.data():
                data1 = [sweep.nodeAddress()], [dataPoint.channelName()], [dataPoint.as_string()]
                if data1[0][0] == 20928 and data1[1][0] == 'ch3':
                    ch3_20928data = data1[2][0]

        for sweep in sweeps:
            data = sweep.data()
            for dataPoint in sweep.data():
                data1 = [sweep.nodeAddress()], [dataPoint.channelName()], [dataPoint.as_string()]
                if data1[0][0] == 20928 and data1[1][0] == 'ch4':
                    ch4_20928data = data1[2][0]

        for sweep in sweeps:
            data = sweep.data()
            for dataPoint in sweep.data():
                data1 = [sweep.nodeAddress()], [dataPoint.channelName()], [dataPoint.as_string()]
                if data1[0][0] == 20928 and data1[1][0] == 'ch5':
                    ch5_20928data = data1[2][0]

## Put data on screen for ch1
ch1_20928da = IntVar()
label2 = Label(root, textvariable = ch1_20928da,padx=2, pady=5, bg = "green", bd = 11, fg = "black", font = ("Castellar",14,"bold"))
label2.place(x=20, y=330)

## retrieve ch1 and bind data from queue
def ch1_20928d(event):
    ch1_20928da.set(commQueue.get())
root.bind('<<ch1_20928>>', ch1_20928d)

## Put data on screen for ch2
ch2_20928da = IntVar()
label3 = Label(root, textvariable = ch2_20928da, padx=2, pady=5, bg = "green", bd = 11, fg = "black", font = ("Castellar",14,"bold"))
label3.place(x=250, y=330)

## retrieve ch2 and bind data from queue
def ch2_20928d(event):
    ch2_20928da.set(commQueue.get())
root.bind('<<ch2_20928>>', ch2_20928d)

## Put data on screen for ch3
ch3_20928da = IntVar()
label4 = Label(root, textvariable = ch3_20928da, padx=2, pady=5 , bg = "green", bd = 11, fg = "black", font = ("Castellar",14,"bold"))
label4.place(x=500, y=330)

## retrieve ch3 and bind data from queue
def ch3_20928d(event):
    ch3_20928da.set(commQueue.get())
root.bind('<<ch3_20928>>', ch3_20928d)

## Put data on screen for ch4
ch4_20928da = IntVar()
label5 = Label(root, textvariable = ch4_20928da, padx=2, pady=5 , bg = "green", bd = 11, fg = "black", font = ("Castellar",14,"bold"))
label5.place(x=750, y=330)

## retrieve ch4 and bind data from queue
def ch4_20928d(event):
    ch4_20928da.set(commQueue.get())
root.bind('<<ch4_20928>>', ch4_20928d)

## Put data on screen for ch5
ch5_20928da = IntVar()
label6 = Label(root, textvariable = ch5_20928da, padx=2, pady=5 , bg = "green", bd = 11, fg = "black", font = ("Castellar",14,"bold"))
label6.place(x=950, y=330)

## Put data on screen for ch5
def ch5_20928d(event):
    ch5_20928da.set(commQueue.get())
root.bind('<<ch5_20928>>', ch5_20928d)

## create and start thread
def start_th():
    ## Run the thread and the GUI main loop
    thread=threading.Thread(target = timeThread)
    thread.start()

root.mainloop()

ccd50, 2022-10-27:
in addition I have configured the node using SensorConnect to log and transmit, and sampling to 256Hz

ccd50, 2022-10-28:
Another note after some minor adjustments to the code I am able to run this with a **WSDA 200** but I am still getting the same issues that I have in my original post _"WSDA-2000 Enable Beacon command has failed"_. When running this same code with a **WSDA 2000**. 

Is it possible this is related to the two deferent methods of communications (WSDA 2000 tcp/ip bus Vs. WSDA 200 serial bus)? Do I need to change the buffer size for the tcp/ip?

**note:** I am using a USB cable to connect the WSDA 2000 to the computer 

is it possible that I need to connect the WSDA 2000 using an ethernet connection

I am looking for any help with this issue.

Thanks

msclissa, 2022-10-31:
I'm glad you got this working with the WSDA 200! The WSDA 2000 should behave in much the same way, communication should work fine when connected via USB (virtual ethernet).

I'm not sure exactly what may be going wrong, but I have a few suggestions to try to see if they help stabilize things. Below I've pared down your provided code to include only the device interactions via MSCL for easier reference. From there I've made a couple code changes that are also described here:
- in `stop()`
  - I would recommend disabling the beacon on the base station within this function after setting the node(s) to idle, it just helps to set everything back to pre-sampling state.
  - It is important to wait for the `idleStatus` object to report that it is complete. When sampling, nodes listen for a set to idle command at a configurable interval (a few seconds). When you call `setToIdle()` in MSCL, the base station starts sending the idle command to the node repeatedly until it hears it. If you don't wait for the operation to complete and another command is sent to the base station it will interrupt this process and the node may not be set to idle.
- in `timeThread()`
  - You definitely want to move the `network.startSampling()` to outside the while loop - this only needs to be called once when you want to re-initialize sampling. I'm surprised you were not seeing more errors with this. If it's still giving you trouble after moving it, I would additionally try completely setting up the network right at the beginning of the `timeThread` function (move the commented `# setup network` section to directly before `network.startSampling()` call). This again just ensures that the state of everything is consistent every time.

Hope this helps, let us know if those changes don't resolve the issue!

```
# Basestation, node
basestation = mscl.BaseStation(mscl.Connection.TcpIp("fd7a:cafa:0eb7:6578:0002:51dc::1", 5000))
node_a = mscl.WirelessNode(20928, basestation)

# setup network
network = mscl.SyncSamplingNetwork(basestation)
network.addNode(node_a)
network.applyConfiguration()

# stop/idle function
def stop():
    idleStatus = node_a.setToIdle()
    
    ## it is important to wait for the idle operation to complete - if interrupted the base station will stop sending the Idle command to the node
    ## IMPORTANT this will run forever if communication with the node cannot be re-established, you may want to add some timeout logic here
    while(not idleStatus.complete())
        pass
    
    ## disable beacon after setting nodes to idle, have seen some weird behavior starting sampling while the beacon was still on
    basestation.disableBeacon()

def timeThread():    
    ## syncronize basestation and start sampling @ 500
    network.startSampling()
    while True:
        sweeps = basestation.getData(500)
        ## do stuff

## create and start thread
def start_th():
    ## Run the thread and the GUI main loop
    thread=threading.Thread(target = timeThread)
    thread.start()
```

ccd50, 2022-11-01:
Thank you for the suggestions.
However, after making all the recommended changes including moving the network.startSampling() outside of the loop and completely setting up the network right at the beginning of the time Thread function. I know have no data streaming from the thread. Is it possible that the tkinter Thread is causing the problem and preventing the sweep of the data?

ccd50, 2022-11-04:
What is the recommended GUI platform with Python to interact with your hardware? For example I am using tanker but, I do know there is a few more out there.

ccd50, 2022-11-04:
Would LabView be an easier path forward? We will have 15 channels streaming at 256Hz each

joabnd7, 2022-11-05:
working on a fix for you to try

closed 2024-10-24 by github-actions[bot]