LORD-MicroStrain/MSCL #302: The ExternalGNSSUpdate command has failed. (Error Code: 4)

Inrixia, 2022-11-09:
I am trying to send an ExternalGNSSUpdate to a 3DM-GX5-45 IMU to initialize its estimation filter (It wont without a GNSS update).
This is what I currently have setup for testing:
```python
node = mscl.InertialNode(mscl.Connection.Serial(args.path, args.baudRate))

gnss_update_data = mscl.ExternalGNSSUpdateData()

gnss_update_data.gpsWeekNumber = 2758
gnss_update_data.gpsTimeOfWeek = 1667956872

gnss_update_data.lattitude = 44.4377
gnss_update_data.northPositionUncertainty = 2.5

gnss_update_data.longitude = -73.1057
gnss_update_data.eastPositionUncertainty = 2.5

gnss_update_data.altitudeAboveWGS84Ellipsoid = 122.9469
gnss_update_data.downPositionUncertainty = 2.5

gnss_update_data.downVelocity = 0
gnss_update_data.downPositionUncertainty = 0.3
gnss_update_data.eastVelocity = 0
gnss_update_data.eastVelocityUncertainty = 0.3
gnss_update_data.northVelocity = 0
gnss_update_data.northVelocityUncertainty = 0.3

print(node.sendExternalGNSSUpdate(gnss_update_data))
```

And this is the error I am receiving:
```
Traceback (most recent call last):
  File ".\src\lil.py", line 51, in <module>
    print(node.sendExternalGNSSUpdate(gnss_update_data))
  File "C:\Users\HSI 9\OneDrive - DTA\Desktop\LILRepo\src\mscl.py", line 7290, in sendExternalGNSSUpdate
    return _mscl.InertialNode_sendExternalGNSSUpdate(self, gnssUpdateData)
mscl.Error_MipCmdFailed: The ExternalGNSSUpdate command has failed. (Error Code: 4)
```

I cant find any more info on what Error Code 4 is and the values used here work perfectly fine when a external update is sent using SensorConnect.

Could I get some clarification on what's going wrong here?

msclissa, 2022-11-16:
It looks like just a copy/paste issue - you have `downPositionUncertainty` set twice, I suspect the second should be `downVelocityUncertainty`. In my testing it worked once that was updated.

There is also a requirement that the GNSS source be set to External, but it sounds like that was already configured on your device.
```
node.setGNSS_SourceControl(mscl::InertialTypes::GNSS_Source::EXTERNAL_GNSS);
```

Hope this helps, let us know if that doesn't resolve the issue!

Inrixia, 2022-11-19:
Ah! I triple checked but of course missed it haha.

Will make that change and confirm here if it's working.
@glenn-t

glenn-t, 2022-11-24:
This resolved the issue, thanks.

closed 2022-11-24 by Inrixia