LORD-MicroStrain/MSCL #303: Assistance with Installing LabVIEW - MSCL

G-Reyes01, 2022-11-10:
I am reaching out today to see how I would go about installing the LabVIEW MSCL VI Package for use of the G-Link-200-8G-OEM within NI LabVIEW. I tried following the instructions on the microstrain website as well as the github repository but have had no luck. Please let me know what it is that I need to do and thank you for your help.

msclissa, 2022-11-16:
What is the behavior you're seeing? Any error messages?
Was the `LORD Sensing - MicroStrain` sub-pallet added to the standard Functions (block diagram) palette?

G-Reyes01, 2022-11-17:
I am actually unable to get the sub-pallet added at all. I am unsure if maybe I am going about the process incorrectly. When I searched in the NI package installer, I was not able to find anything with the terms "MSCL" or "LORD Sensing."

G-Reyes01, 2022-11-17:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Good Morning,

In regards to the error, I actually am not facing any specific errors. Instead, I was unable to even install the NI package as a whole. When using the package installer, I did not see anything with "MSCL" or "LORD Sensing" within the name. I don't know if maybe I am going through the installation process incorrectly or if maybe I am missing a step.

Please let me know what we can do, what potential issues I may be facing and thank you for your help!

Best Regards,
German Reyes

German Reyes Jr.
Graduate Student
Mechanical Engineering
Research Assistant - Structural Dyancmics and Acoustic Systems Lab
University of Massachusetts Lowell
Email: ***@***.******@***.***>
Phone: 978-885-3819

cpallan, 2022-12-15:
You have to download the .vip file and install it with the JKI VI Package Manger (not the NI Package Manager) for the LV version you are using.
[image]

G-Reyes01, 2022-12-16:
Good Morning,

When I was searching for MSCL, I initially was looking through the JKI VI Package Manager. I had a screen which appeared the same as the one as the attached image. However, nothing appeared as you have when I searched for “lord.”

Could this be due to my version of LabView being a 2017 installation? I see that the version you are using in your search is for a 2020 LabView installation.

Please let me know what else you think it can be and I hope to hear from you soon.

Best,
German Reyes

German Reyes Jr.
Graduate Student
Mechanical Engineering
Research Assistant – Structural Dynamics and Acoustic Systems Lab
University of Massachusetts Lowell
Email: ***@***.******@***.***>
Phone: 978-885-3819

[image]<https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F25420001%2F207927956-c06a347b-918d-4c93-b799-05b842fdf459.png&data=05%7C01%7Cgerman_reyes%40student.uml.edu%7C45e03bca8c934e0a2c4108dadec1ec0b%7C4c25b8a617f746f983f054734ab81fb1%7C0%7C0%7C638067221720079464%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=00ggHvhTY4cjE5xVkAiKPAq1Yo6N8p9O6MksxZXV6Do%3D&reserved=0>

cpallan, 2022-12-16:
You have to select LV2017 and install it when that is selected in the VI Package Manager. It should then show up when you search for Lord when you have 2017 selected. Note that the latest package on the Lord git site is for 32-bit labview, not 64.

G-Reyes01, 2022-12-29:
Good Morning,

Is there a version for a 64-bit build available? If so, how would I be able to go about installing this version?

Best Regards,
German Reyes

German Reyes Jr.
Graduate Student
Mechanical Engineering
Research Assistant - Structural Dynamics and Acoustic Systems Lab
University of Massachusetts Lowell
Email: ***@***.******@***.***>
Phone: 978-885-3819

cpallan, 2022-12-29:
This is the only 64-bit Lord library on the git repo. It is downrev from
the latest 32-bit version, but seems to work okay. Lord says that they
don't support it.

https://github.com/LORD-MicroStrain/LabVIEW-MSCL/releases/tag/v0.6.1.10

closed 2024-10-24 by github-actions[bot]