LORD-MicroStrain/MSCL #305: Connect to multiple nodes

dheazkelvin, 2022-11-15:
Can I have examples for connecting to multiple nodes and how MSCL parsing data for multiple nodes?

dheazkelvin, 2022-11-15:
@msclissa

dheazkelvin, 2022-11-16:
And how we separate data from node1 and node2?

joabnd7, 2022-11-30:
Assuming Python:

Begin with a list of node IDs (can find these and set them up using Sensor Connect),
`node_ids = [10000,20000]`
Then setup your connection and  base station 
`connection = tcpip_connect(tcpip_address,server_port)`
`base_station = mscl.BaseStation(connection)`
Then connect nodes to your base station
`nodes = connect_nodes(node_ids,base_station)`
where the function connect_nodes looks something like this (ignore my logger function):
`def connect_nodes(node_ids,basestation,logger):	

	nodes = []
	msg('Building nodes....',logger)
	for node_id in node_ids:
		msg('Attempting to connect Node: {}'.format(node_id)+' to BaseStation...',logger)
		
		try:
			#create wireless node connection to basestation
			node = create_node(basestation,node_id,logger)
			#test node status
			base_strength , node_strength = test_node_status(node,node_id,logger)
			msg('Node:{}'.format(node_id)+'connected, signal strength: node={}'.format(node_strength)+' Base={}'.format(base_strength),logger)
			#add to list of connected nodes
			nodes.append(node)
		
		except:
			msg('Node failed to connect:{}'.format(node_id), logger)

	return nodes `

As for your second question, the trick there is to bring in your sweeps data into a Pandas Data Frame. One of the columns of that data will have the node ID. Sort the data frame using the node ids. At that point you are beyond the Lord software and are into your own specific application. I use this function after getting all the sweeps (see docs):
`df_node = node_extract(df,node)`
where this helper function is 
`def node_extract(df,node):
	'''
	Helper function to extract node/chan data from a dataframe
	where data is arriving in sweeps. Extracts all data for a 
	given chan on a specific node
	'''
	df_node = df[df['node'] == node]	
	return df_node`