LORD-MicroStrain/MSCL #310: Crashing using MSCL v0.6.1.4 (latest)

cpallan, 2022-12-12:
I am using Labview 32-bit v2020sp1 with MSCL v0.6.1.4 and am seeing intermittent crashes. I get data, but can't seem to fix the crashing. My program doesn't have a memory leak.

How can I find the root cause of the problem so I can correct it.

Thanks,
Charlie

joabnd7, 2022-12-15:
not sure if this will help you but using a pure Python based approach has uncovered some things with how this hardware works. I have very little confidence that the Labview driver can deal with all of the strange things that can happen. But to at least try and be helpful note that the SamplingType can return 0 or 7 which are garbage and you only want to accept sweeps when the `sweep.samplingType() ` reports 2.

joabnd7, 2022-12-15:
the other thing to try is to put in some time delays so that you are not asking for more sweeps while the base station is still writing out sweeps. This can also cause a crash. You can then reduce the time delays later until you find an optimum.

cpallan, 2022-12-15:
Hi joabnd7,
Thank you VERY much for getting back to me with this information. I have added a check to my code to verify that the samplingType is 2 before passing the information on to other vis.

cpallan, 2022-12-15:
Regarding the asking for more sweeps while the base station is still writing sweeps, I have the Basestation Get Data.vi before the MSCL-Utility Parse DataSweep.vi in a timed loop that I run at 200mS. Do you think that this is a sufficient wait time and methodology? 

[image]

cpallan, 2022-12-16:
Hi, the code is still crashing. Btw, I'm using 2 base stations, but have
changed all the code to reentrant, whit the exception of the base station
vis which I missed. I have tested with only one base station, and so far,
no crash. I'll try 2 again next.

If it still crashes, I'll test your time delay idea.

Thanks again for the help! Very much appreciated, I've been thrashing
around on this problem for 3 weeks now...

joabnd7, 2022-12-16:
I think you are on the right track, and I am glad to try and help out. I am using a pure Python approach and another thing you might want to try is to limit the number of sweeps you acquire each time your loop executes. The docs explain that when you call 
`basestation.getData(timeout,num_sweeps)`
you can request a certain limited number of sweeps along with the timeout. Try a timeout of 0 and 100 sweeps and see if that works any better. The reason I say timeout = 0 is that is the default and the docs are confusing whether this is actually used as a timeout or as a time delay before starting to acquire sweeps. But I know for sure if you do not limit the number of sweeps and put the acquire in a loop structure for real time monitoring, it will eventually choke.

cpallan, 2022-12-17:
Hi joabnd7,
I have tried to put a limit of 100 sweeps, but it slows down data reception
- typically the nodes return 120-150 samples, so I think that by limiting
it to 100, the node no longer keeps up to real time. What do you think of
acquiring 200 samples at a time? I guess that this is basically the same as
putting in 0 and getting 'all' samples...

I have tried a time delay before BaseStation GetData.vi and between it and
the Parse DataSweeps vi. The system was still crashing, but I have new
news: I changed a labview structure "timed loop" to a regular while loop,
and the system stopped crashing. It just stops reading data from the Lord
nodes. The other piece of information is that I have traced the problem
down to my setup of using 2 base stations simultaneously; if I use only
one, the system keeps receiving data with crashing or stopping (with your
settings). So maybe this is a new problem where the .net library cannot be
called twice simultaneously. I have stopped that from happening with a
semaphore, but that made no difference - it still stopped receiving data,
which surprised me.

Thanks again for your help! VERY much appreciated!

joabnd7, 2022-12-17:
I am not sure I agree that asking for 200 sweeps is the same as using the default =0 for number of sweeps because of the circular buffer, but you know your setup better than I do. We are also using two devices - in our case a basestation and an IMU. The way I am handling that in Python is by running each device using a separate thread. Both threads call the MSCL library but they don't know each other are running. I believe there is a way to do this same kind of thing in Labview. One obvious question for you - is why two base stations, since one can handle a large number of nodes... There is a command to get the % of bandwidth being used by the base station. Are you using all the bandwidth of one base station and so you need another one? To get the bandwidth - after setting up a SynchSamplingNetwork I use the command
 `network.percentBandwidth() `
 to get the bandwidth being used. It was helpful to run this command and add one node at a time to the SynchSamplingNetwork and watch how each node changed the bandwidth. You can also get crashes if the bandwidth reaches above about 90%.... on the basestation. This is a function of the sampling rate and number of channels. If you are used to using NI and Labview with high speed DAQ cards like PXI etc... this system is quite different being wireless - it has bandwidth limitations way lower than you might be used to using.

joabnd7, 2022-12-18:
Here is a link to get you started on multithreading in Labview in case it is new to you:
[https://www.ni.com/en-us/support/documentation/supplemental/15/multithreading-in-labview-real-time.html]
I believe that since you got one basestation to work ok, that getting two using multithreading should solve your problem.

joabnd7, 2022-12-18:
Last thing I will mention is also a possible issue where the addressing for the basestations is not unique, but I am pretty certain you probably already checked that.  In Python this would be 
`station1 = mscl.BaseStation(connection1)`
`station2 = mscl.BaseStation(connection2)`

joabnd7, 2022-12-18:
One other thought - the latest version of MSCL is 6.5 - make sure you are using the latest version

cpallan, 2022-12-18:
I did not know that there was a later version of the .net library. When I try to use it, it throws an error: 

[image]

cpallan, 2022-12-18:
Am I downloading the correct file: 
[image]

joabnd7, 2022-12-18:
Well clearly that was bad advice on my part. Only idea there is to make sure the drivers you select are for 32 bit if you are using 32 bit Labview - or 64 bit if you are using 64 bit Labview. I think you said you were using 64bit. I am hoping that someone from Parker can chime into this thread about that particular error. Just a reminder that I am a kind stranger trying to help - and do not work for them... If it makes you feel any better we are also having issues getting this all working on a complex system. I would return to the drivers you were using that worked ok with a single base station and try out the multithreading for two base stations.

cpallan, 2022-12-18:
Hey joabnd7,
Well, first things first: I made a mistake and did not copy both .net files to the x86 directory. 

> The reason for two base stations is the bandwidth. I'm well over 50% on both base stations according to Sensor Connect.
> Sensor connect runs both base stations fine. This leads me to believe that it is actually possible.
> I am specifying two different IP addresses, one for each base station, but am not specifying the port. I checked in Sensor connect and they are using different IP addresses, but the same port number: 5000. I am not sure how to specify different ports in Labview, but I will try to figure it out, just to see if it makes a difference.
> I am going to study the multitasking document from the link you sent; it describes execution systems in Labview, which I have always wondered about and have never understood. So thank you for that.
> I am using 32-bit labview because I found it to be more stable running the Lord product (less crashes.)
> I am testing the x86 driver from MSCL v6.5, with 32-bit Labview.
> I will return to v6.1.4 as it is marked as the "latest" on the Labview git page.
> I have run multiple threads; this appears to be where the problem lies (I think.)
> Thank you, thank you, thank you, for taking the time to help me on this. I am in a real crunch (as we all seem to be always, in this business), and understand that you do not work for Lord, and therefore must be a very kind person...

[image]

joabnd7, 2022-12-18:
Sounds like you are getting close. The other idea I had for you was to write a small Python code that does the multi-threading for you and have Labview just run the Python script at the start-up and then ingest the data output from both of those threads into the rest of your Labview GUI. But I do believe there is a way in Labview to do exactly that same thing. 
see Figure 9.5 where in the VI properties itself you can assign
`https://web.physics.utah.edu/~jui/3620-6620/supp/LV_Adv_Prog/2049ch09.pdf`

cpallan, 2022-12-19:
I have fixed the "out of memory" error. I had removed a close reference in a loop where I misunderstood the code and thought that it was not needed. This caused the program to exceed the maximum open references of 1M+... I'm testing now, and so far, so good. I have no confidence that will be the case if I try to run overnight... but my fingers and toes are crossed.

That is a very "outside the box" thinking to write it in Python and call that from Labview. I believe that I'm already doing this same function doing multitasking in Labview, but if it keeps failing, that does give me an alternate path. Thank you for that idea.

I am trying to understand the VI properties from that article you sent. I'm still a bit confused, but getting there.

cpallan, 2022-12-19:
The VI properties article you sent got me out of the woods on this problem - it's fixed. I changed the VI's "Preferred Acquisition System" from "Same as caller" to "data acquisition". As far as I can tell, this changes the thread on which the Lord code runs away from the user interface thread, and on to a separate thread. 

Are you with University of Utah? I noticed that was on the link for the VI properties was from there. I appreciate so much your help with this problem. I would have not been able to solve it without your help. So what's your favorite vice? I, personally, am a  Scotch drinker and would like to send you a nice bottle of that or something else of your choice. If you don't drink, I would like to send you another gift to say thank you for your time.

This project I'm finishing here is for Joby Aviation for a NASA wind tunnel test that we are going to perform. I have a window to finish that is closing, so getting this problem fixed is a HUGE relief. Thank you again.

joabnd7, 2022-12-19:
Awesome!! I found that article for you just digging around- it is a CRC Press book chapter and the CRC always has good stuff. Glad to hear that worked. I'll send you an email.

closed 2022-12-22 by cpallan