LORD-MicroStrain/MSCL #312: How to handle robustness of creating a serial connection for the WSDA-200-USB Error: Failed to read the Firmware Version (EEPROM 108) from the BaseStation

ybabs, 2022-12-22:
Hello, we have two different gateways sampling collecting data from various nodes. One is the WSDA-Base-LXRS while the other is a WSDA-200-USB. 

We create a serial for connection and a basestation object like this:

```
LOG_INFO << "Setting up WSDA -200Network:" << m_PortName ;
    try{
        m_Connection = mscl::Connection::Serial(m_PortName, 3000000);
        m_BaseStation = std::make_unique<mscl::BaseStation>(m_Connection);
        m_Network = std::make_unique<mscl::SyncSamplingNetwork>(*m_BaseStation);
    }
    catch(mscl::Error_Connection& e) {
        LOG_ERROR <<  "Error: " << e.what() << " (" << e.code() << ")";
    }
    catch (mscl::Error& e)
    {
        LOG_ERROR <<  "Error: " << e.what() ;
    }

```
Only difference is the LXRS-Base runs at 115200. The problem we have is ever so often the WSDA-200-USB will return a `Error: Failed to read the Firmware Version (EEPROM 108) from the BaseStation` and you basically have to run the program two or three times before you get a connection. Does anyone know why this happens?

My original theory was maybe when you shut the program down, the basestation is not exiting gracefully as the beacon is still blinking blue. I've tried adding this to the destructor on shutdown but it doesn't seem to have done anything. 

```
    m_BaseStation->broadcastSetToIdle();
    while(!m_BaseStation->broadcastSetToIdle().complete())
    {
    }
    m_Connection.disconnect();
    m_BaseStation->disableBeacon();
```
We don't have this issue when we use the SensorConnect Application. Can anyone help figure out what is going on?

joabnd7, 2022-12-22:
My initial thought is that you need to run each of the base stations in a separate thread using multithreading. This is what I and one other person on here have found in Python and also Labview. I hope that helps  you.

ybabs, 2022-12-22:
Hi @joabnd7 , I mostly wrote this code snippet for brevity. Each basestation runs in separate threads and writes it data to separate CSV files. It'sa bit of a weird one as I never experience the issue with the Base LXRS. It's always the WSDA-200 every single time

joabnd7, 2022-12-22:
Oh ok,.. makes sense. Was just an initial thought. Have you tried replacing the cable?

ybabs, 2022-12-22:
I've got it connected directly to the Pi3 as shown below

[image]

joabnd7, 2022-12-23:
Well that rules out a bad cable... :)  One suggestion with setups like this is to try and use a powered USB hub to relieve the power supply task from the Pi3 to a separate power (perhaps more stable) source. So you would have a USB cable from the Pi3 to a powered USB hub output. And then connect the WSDA-200 to an input port on the powered USB hub. Not sure this will solve your issue since it all works on SensorConnect... but it is so easy to try.

ybabs, 2022-12-23:
I'll give this a go tomorrow thanks. I did try to turn down the transmit power on the gateway as well to reduce power requirements but that didn't really do much

joabnd7, 2022-12-23:
one last thought - did you try changing the order of the disconnect and disable beacon signal in your shutdown script? 
i.e. once the connection is shutdown you can't send the disable beacon signal

closed 2023-11-14 by ybabs