LORD-MicroStrain/MSCL #318: Connection error using WSDA 200 usb with raspberry pi 4 b (64bit).

piyush0017, 2023-01-30:
HI,
I am trying to connect with sg-link nodes using WSDA 200 usb on raspberry pi 4b 8gb 64 bit. Which has Ubuntu 22.04 LTS Desktop installed on it. 
I am using this python code to connect(which works fine on windows with COM).

`import sys
sys.path.append('/usr/share/python3-mscl')
import mscl
#import numpy as np
#import openpyxl
#import os
#COM_PORT = os.path.realpath("/dev/ttyUSB0") 
try:
    #create the connection object with port and baud rate
    connection = mscl.Connection.Serial("/dev/ttyUSB0", 3000000)

    #create the BaseStation, passing in the connection
    basestation = mscl.BaseStation(connection)

    node = mscl.WirelessNode(50***, basestation)
    node1 = mscl.WirelessNode(40***, basestation)
    # ping the Node
    #response = node.ping()

    # if the ping response was a success
    #if response.success():
        #print(response.baseRssi()) # the BaseStation RSSI
        #print(response.nodeRssi()) # the Node RSSI

    # create a SyncSamplingNetwork object, giving it the BaseStation that will be the master BaseStation for the network
    network = mscl.SyncSamplingNetwork(basestation)

    # add a WirelessNode to the network.
    # Note: The Node must already be configured for Sync Sampling before adding to the network, or else Error_InvalidNodeConfig will be thrown.
    print('Adding node1 to network..',network.addNode(node))
    print('Done')

    print('Adding node2 to network..',network.addNode(node1))
    print("done")

    print("network ok:",network.ok())                # check if the network status is ok
    #network.lossless(true)      # enable Lossless for the network
    #network.percentBandwidth()  # get the total percent of bandwidth of the network

    # apply the network configuration to every node in the network
    print("applying net conf",network.applyConfiguration())
    print("done.")

    # start all the nodes in the network sampling.
    print("strating the network..",network.startSampling())
    print("done.")

    
except Exception as e:
    print ("Error:",e)  `

But when I run this I get "Connection error" 

What could be the possible reason for it. 

I was not able to install cp210x driver from the page, but the device is been identified by pi and lsmod shows cp210x already installed. also grep tty shows following,
[    0.974583] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 19, base_baud = 0) is a PL011 rev2
[ 3272.969340] usb 1-1.3: cp210x converter now attached to ttyUSB0

I got similar error while using linux virtalbox on windows where i was able to install the provided cp210x driver.  

regards
piyush

joabnd7, 2023-02-04:
have you tried a different baud rate?

juanvillacres, 2023-02-24:
Use sudo as suggested in:

#https://github.com/LORD-MicroStrain/MSCL/issues/107#issuecomment-589014755

piyush0017, 2023-03-31:
> have you tried a different baud rate? base baud = 0 is a clue to me that the baudrate might be the issue.

Hi joabnd7,

Yes I had tried using different baud rates and even no baud rate but was not able to connect.

closed 2024-10-24 by github-actions[bot]