LORD-MicroStrain/MSCL #319: SensorConnect.exe AppHangB1

VincentMassingham, 2023-02-04:
Whilst using MSCL to interface an accelerometer to a SoftPLC running in a PC, after some time the SoftPLC crashed.
 
When I interrogated the PC I noticed on the Win Event Viewer (Windows Logs\Applications) it was generating ‘Sensor Connect’ events (information) every 20min ? It had not done this before. Another system which is running OK does not generate any ‘Sensor Connect’ events. 
 
Below is the event detail:
 
Fault Bucket, type 0
Event Name: AppHangB1
Response: Not Available
Cab Id: 0
 
Problem signature:
P1: SensorConnect.exe
P2: 0.0.0.0
P3: 606b914b
P4: d5f7
P5: 134217728 
 
I ran the ‘Sensor Connect’ software, it worked fine and could read X Y Z values. We therefore know the actual base node & accelerometer are ok. The problem lies in the C++ interface to our SoftPLC using your MSCL library.

The SoftPLC was restarted but the above event continues to be posted every 15min.

Can anybody help?

joabnd7, 2023-02-04:
I will do my best to help here until you hear back from Lord/Parker ( I do not work for them). It is my understanding that you cannot have a sensorconnect.exe or SensorConnect app running while you are also using the MSCL library with C++ (or Python). I believe that what you might have seen is some leftover Sensor Connect process was running (maybe even in the background when you were attempting to use the MSCL libraries. You can use Sensor Connect to setup your channels bu then it must be dully exited before trying to use the MSCL library. Good Luck!

VincentMassingham, 2023-02-28:
I have uninstalled SensorConnect, yet I still get the events detailed above. Is there a special method other than win remove programmes, to totally uninstall SensorConnect.

closed 2024-10-24 by github-actions[bot]