LORD-MicroStrain/MSCL #341: Configure sensor in code

saytesla, 2023-06-08:
[image]
I'm trying to configure my sensor in code. Above is what I have currently as my configurations. 

I'm having trouble figuring out how I can set config for
Set m/sec^2

Set low pas filter

Set frequency of sensor and basestation?

msclissa, 2023-06-21:
I've included examples of how to set the low-pass filter and frequency below. The examples are in C++ but the interface is the same for Python - let us know if you run into any trouble converting it.

I'll follow-up tomorrow with how to configure the unit. The calibration slope and offset need to be changed to output in the desired unit. Those values are calibrated in-house so I want to properly advise you around maintaining the original values and restoration options if they do end up being cleared.

**Low-pass filter**
```
// low-pass filter is configured for chs 1, 2, 3 together
mscl::ChannelMask chmask;
chmask.enable(1);
chmask.enable(2);
chmask.enable(3);

// get a list of supported low-pass filters
// you could use this to select the closest supported option to a target
// but with the one node type probably easiest to just directly set the desired hz value that you know is supported
mscl::WirelessTypes::Filters supportedLowPassFilters = node.features().lowPassFilters();

// set in config object to be applied
config.lowPassFilter(chmask, mscl::WirelessTypes::Filter::filter_209hz);
```

**Frequency**
```
// check current frequency first (don't bother changing if correct)
baseStation.frequency();
node.frequency(); // not needed if prior successful communication - node would need to be on same freq to communicate

mscl::WirelessTypes::Frequency targetFrequency = mscl::WirelessTypes::Frequency::freq_12;

// Important: node must be changed before base station or communication will be lost
node.changeFrequency(targetFrequency);

// once node frequency is set, you can change the base station frequency
baseStation.changeFrequency(targetFrequency);

// check we didn't lose communication with the node
mscl::PingResponse pingRes = node.ping();
if (pingRes.success())
{
    // communication successful
}
```

saytesla, 2023-06-21:
Can we get this in python?

msclissa, 2023-06-22:
Is there something specific that's not working when you try to convert it to Python?

For changing the output unit from g to m/s^2 there are two properties that need to be set in the config:
- unit indicator
- calibration coefficients: slope and offset

Example code below.

The calibration coefficients tell the device how to convert the raw voltage output to a meaningful value in the desired units. If you have the device configured to output calibrated float data the calibration happens prior to the data being transmitted. It looks like you currently have the output data format configuring to `dataFormat_raw_int16` - with this data format you will need to read the calibration coefficients from the device while it's idle and apply them to output data manually:
```
calibrated data = (output data * slope) + offset
```
If you want the device to output data that is already converted to the correct units you should set the output data format to `dataFormat_cal_float`.

**Important**: G-Links are calibrated here prior to being shipped. The calibration coefficients (slope and offset) are set specifically for each node such that the raw voltage can be converted accurately to g. To change the output unit the pre-calibrated slope and offset need to be adjusted.

The original calibrated values (g) are recorded for each channel on the calibration report that was sent with each node. We also keep a copy of this information and the original node configuration here that we can send you in the event these values are accidentally cleared.

**Configure Unit**
```
// set units and calibration coefficients for each channel
const float G_TO_MS2 = 9.80665f;
uint8_t unitChs[] = { 1, 2, 3 };
for (uint8_t ch : unitChs)
{
    mscl::ChannelMask c;
    c.enable(ch);

    // check current unit, only change if g
    // if currently something other than g or m/s2 the conversion factor will need to be different
    mscl::WirelessTypes::CalCoef_Unit currentUnit = node.getUnit(c);
    if (currentUnit != mscl::WirelessTypes::CalCoef_Unit::unit_accel_g)
    {
        // skip conversion for this channel
        continue;
    }

    // channels are calibrated in-house
    // need to read off the existing calibration (g) and convert it to desired unit (m/s^2)
    mscl::LinearEquation eq_g = node.getLinearEquation(c);
    mscl::LinearEquation eq_ms2;
    eq_ms2.slope(G_TO_MS2 * eq_g.slope());
    eq_ms2.offset(G_TO_MS2 * eq_g.offset());
            
    // set calibration coefficients in config object to be applied
    config.linearEquation(c, eq_ms2);

    // set unit in config object to be applied
    config.unit(c, mscl::WirelessTypes::CalCoef_Unit::unit_accel_mPerSec2);
}
```

closed 2023-07-20 by msclissa