LORD-MicroStrain/MSCL #344: get the NMEA Data

365devpark, 2023-06-20:
Hello.
I have a question during program production using 3DM-GQ7 sensor and MSCL.

My goal is to get the NMEA Data from the sensor, then send it to the NTRIP server to receive the calibration value, and then send it back to the sensor.
What are some ways to get NMEA data from a sensor here? I'd appreciate it if you could let me know.

I'll be waiting for the reply.

msclissa, 2023-06-26:
The GQ7 can be configured to output various NMEA sentences over the Main port. I've included some code below how to configure this via MSCL, but it's probably easiest to use SensorConnect to configure the NMEA output. If it needs to be done via MSCL, consider configuring once in SensorConnect and then writing the MSCL code to have the same options - it's just easier to see what all is available and required. NMEA output configuration can be found in SensorConnect under the **NMEA Message Format** tab on the **Sampling** screen.

More details about the possible NMEA outputs can be found in the GQ7 User Manual here: [Additional Features -> NMEA Output](https://mip-documentation-test.s3.ca-central-1.amazonaws.com/GQ7+Online/user_manual_content/additional_features/NMEA%20Output.htm)

NMEA data can be output on its own or alongside MIP data (standard data output for our devices). For initial setup and testing I would recommend disabling MIP data output as it will be easier to tell that the NMEA output is configured correctly. If you're only using the NMEA output, there is no reason to have MIP output enabled. I've included an example below for disabling all MIP data output.

It is important to note that **setting the device to idle will pause both NMEA and MIP data output**. It's a good idea to always send the resume command before trying to read either type of data from the device.

**There is not currently a NMEA parser implementation in MSCL** but I've included a very basic parser along with an example of how to read off the NMEA data from an MSCL connection object. If the device is already configured and streaming, you don't even need to create an `InertialNode` object, you can just enable `debugMode` and read the raw bytes directly from the `Connection` object.

I hope this helps! Let us know if you have any questions!

**Configure NMEA Output**
```
// read existing NMEA config to add onto it instead of clearing/overwriting
mscl::NmeaMessageFormats nmeaMessageFormats = device.getNmeaMessageFormat();

// create new NMEA format entry
mscl::NmeaMessageFormat nmeaFormat;
nmeaFormat.sentenceType(mscl::NmeaMessageFormat::SentenceType::GGA);
nmeaFormat.talkerId(mscl::NmeaMessageFormat::Talker::GPS);
nmeaFormat.sourceDataClass(mscl::MipTypes::DataClass::CLASS_GNSS1, device.getDataRateBase(mscl::MipTypes::DataClass::CLASS_GNSS1));
nmeaFormat.sampleRate(mscl::SampleRate::Hertz(1));

// add new format to existing list
nmeaMessageFormats.push_back(nmeaFormat);

// set NMEA message format on device
device.setNmeaMessageFormat(nmeaMessageFormats);
```

**Disable MIP Data Output**
This will configure the device so that MIP data will not be output even once the resume command is sent. It does not clear the configured MIP data message format.
```
mscl::MipTypes::MipDataClasses dataClasses = {
    mscl::MipTypes::DataClass::CLASS_AHRS_IMU,
    mscl::MipTypes::DataClass::CLASS_ESTFILTER,
    mscl::MipTypes::DataClass::CLASS_GNSS,
    mscl::MipTypes::DataClass::CLASS_GNSS1,
    mscl::MipTypes::DataClass::CLASS_GNSS2,
    mscl::MipTypes::DataClass::CLASS_GNSS3,
    mscl::MipTypes::DataClass::CLASS_GNSS4,
    mscl::MipTypes::DataClass::CLASS_GNSS5,
    mscl::MipTypes::DataClass::CLASS_SYSTEM
};

// loop through all possible data classes
// disable streaming if supported
for (mscl::MipTypes::DataClass dataClass : dataClasses)
{
    if (device.features().supportsCategory(dataClass))
    {
        // disable streaming for data class
        // change false param to true to re-enable
        device.enableDataStream(dataClass, false);
    }
}
```

**Read Data and Basic NMEA Parsing**
This parser should work alright even if the device is outputting both MIP and NMEA but it is not very robust and has not been thoroughly tested. It may need adjustment.
```
// resume
// when device is idle it prevents NMEA as well as MIP data output
device.resume();

// setup connection
// debug mode allows raw traffic to be buffered and read back out from the Connection object
connection.debugMode(true);

[... 60 lines cut ...]
                else
                {
                    checksumCountdown = NMEA_FOOTER_LENGTH;
                }
            }
            // if end byte read, detect if checksum countdown completed and clear current
            else if (byte == NMEA_END)
            {
                // not valid if checksum countdown isn't exactly 0 (four bytes since checksum indicator)
                if (checksumCountdown == 0)
                {
                    std::cout << "Found NMEA: " << currentNmeaSentence << endl;

                    ////////////////////////////////////////////////////////
                    // insert logic to handle complete NMEA sentence here //
                    ////////////////////////////////////////////////////////
                }

                // clear current NMEA sentence
                currentNmeaSentence = "";
            }
        }
    }
}
```

365devpark, 2023-06-28:
First of all, thank you for your kind reply!

I was able to receive the NMEA data well when I used the example you told me.

Now, in order to proceed to the next step, I need to receive RTCM data from the NTRIP server and send it to the sensor, but can you tell me how to do this?

After checking the MSCL documentation, we found a function called sendRawRTCM_2_3Message, but not a function that supports RTCM 3.2.

I'd appreciate it if you could answer!

msclissa, 2023-06-29:
Apologies for the confusion, I'm not actually sure which device supports that `sendRawRTCM` function/command. Luckily RTCM data can easily be written direct to a `Connection` object to the Aux port. I have included example code of what that looks like below.

RTCM data must be passed into the GQ7 via the Aux port. The GQ7 can also output NMEA over the Aux port, but it is not configurable like the NMEA output over the Main port.

**Main Port Functions/Outputs**
- Setup Main port NMEA output
- Enable Aux port NMEA output, RTCM processing
- MIP data output and configuration

**Aux Port Functions/Outputs**
- NMEA output (details based on my observations, need to confirm exact behavior with product expert)
  - GGA only
  - Data source: estimation filter
  - Data rate: ~0.5 - 1 hz
- RTCM input processing
- No device configuration

If the non-configurable Aux port NMEA output is sufficient for what you need it may be easier to interact with just the Aux port for NMEA out/RTCM in, but you can certainly use the Main port as your NMEA source as well.

The **Read Data and Basic NMEA Parsing** example in my previous response will also work for either port, just make sure the `connection` object is open to the correct target (Main or Aux).

You should not make an `InertialNode` object with an Aux port `Connection` object - most functions will be unavailable over the Aux port. All required `InertialNode` functionality needs to be accessed through the Main port.

Hope this helps - let me know if anything needs clarification!

**Enable Aux Port NMEA Output & RTCM Processing (via Main port)**
```
// open connection to GQ7 Main port and make InertialNode object
// for:
//  - configurable NMEA output
//  - enable/disable Aux port NMEA output
mscl::Connection mainPort = mscl::Connection::Serial("COM24", 115200);
mscl::InertialNode device = mscl::InertialNode(mainPort);

// enable Aux port NMEA output (non-configurable)
device.enableRtk(true);

// save settings as startup so it does not need to be re-configured on power-on
device.saveSettingsAsStartup();
```

**Write RTCM Data to Aux Port**
There are two format options for writing data directly to a `Connection` object: a vector of uint8 and a string. I suspect you'll have your RTCM data in a byte array or vector, but I wanted to include an example for string as well in case it's easier for you.
```
// open connection to GQ7 Aux - interact directly with the Connection object
// for:
//  - non-configurable NMEA output (GGA)
//  - RTCM input
mscl::Connection auxPort = mscl::Connection::Serial("COM4", 115200);

// RTCM data in a uint8 vector format
std::vector<uint8_t> rtcmData;

// write the RTCM data to the AUX port connection
auxPort.write(rtcmData);

// RTCM data in a string format
std::string rtcmData_str;

// write the RTCM string data to the AUX port connection
auxPort.writeStr(rtcmData_str);
```

msclissa, 2023-07-20:
I'm going to go ahead and close this as resolved. Please feel free to re-open if anything needs any additional clarification or you run into any issues! Thanks!

closed 2023-07-20 by msclissa