LORD-MicroStrain/MSCL #366: Will Connection::Tcpip work with my own proxy application?

sebuck, 2023-10-24:
My environment requires a proxy between my external applications and my 3DM-GQ7.  My proxy just needs to read the 3DM-GQ7's raw bytes and resend them over it's own IP/Port.  The external applications do NOT need to send any commands.  Can my external applications use MSCL's Connection::TcpIp to listen to my proxy and will MSCL still properly parse the relayed MIP messages?  Or will MSCL only connect to the actual physical hardware?

msclissa, 2023-10-31:
You should be able to open that connection and send MIP data to be parsed by MSCL. I don't think that I've personally ever tried this kind of setup, but I can't think of any reason data parsing and collection would work differently over this method. Any commands or command responses in the raw bytes will be ignored and any `InertialNode` functions that require sending commands to the device will fail but continuously calling `getDataPackets()` should work as expected.

I would be interested to hear the results if you decide to try this and of course would be happy to help work through any challenges you run in to!

sebuck, 2023-11-01:
Thank you for your quick response!  Our MIPS message stream is simple so I decide to just write my own quick parser instead of using MSCL.  Your documentation is excellent and after digging into it more, I agree that there doesn't appear to be any reason why MSCL's data parsing and collection wouldn't work.  Thank you for confirming this.

closed 2024-10-24 by github-actions[bot]