LORD-MicroStrain/MSCL #368: node connection    (WSDA-200-USB with node adress 57233)

xin-bee-maker, 2023-11-02:
"I am able to establish communication with my gateway, but I am unable to recognize the sensor. I cannot read its information and set its status to sleep mode. Below is my code. I have made some improvements based on previous issues raised by others, but I still cannot achieve it and cannot read any information about the node. However, I can automatically recognize this sensor through the "sensor connect" platform."

msclissa, 2023-11-02:
The code provided does not show any functions that actually attempt to communicate with the node. It may be helpful to know what function is failing and what the error message is.

A couple things to check:
- The node needs to be on the same frequency as the base station in order to communicate. I suspect this is not the issue as you are able to establish communication in SensorConnect. You can check and configure the frequency using MSCL, but it's probably much easier to get this set up through SensorConnect.
- Increase `readWriteRetries` on the node. Just like you have for the base station, it might help to call `node.readWriteRetries(5)` if you are operating in an environment that makes communication challenging. The default value for `WirelessNode` objects is 3.
- The node needs to be in Idle mode to communicate. Nodes have three modes:
  - Sleep: power-saving
  - Sampling: streaming data
  - Idle: actively listening for communication
 
In Sleep and Sampling modes, the only command the node will respond to is Set to Idle. In these modes nodes listen for the Set to Idle at configurable intervals, so the base station continuously sends the command until the node responds. Below is an example of how to call setToIdle and wait for it to resolve:
```
// send Set to Idle command and wait for response
mscl::SetToIdleStatus status = node.setToIdle();
while (!status.complete())
{
    // add timeout logic here so doesn't spin forever if unable to communicate with node
    continue;
}
```

Hopefully this helps, otherwise I will need more information about what exactly is failing and the error message or behavior you are seeing!

xin-bee-maker, 2023-11-06:
Thank you! According to the loop, I can currently set the node into idle mode, but what I want to do is not to put it to idel but instead read some information from it. How can I achieve this? Also, why can't I use a loop to set a timeout, such as "timeout(10000)"? The timeout period has been set very long, but I cannot set the node into idle mode. What factors might cause this problem?

closed 2024-10-24 by github-actions[bot]