LORD-MicroStrain/MSCL #369: Problem using 3DMCV7-AHRS with Raspberry Pi 3B+

gabriele-bruzzone, 2023-11-07:
I'm trying to use a 3DMCV7-AHRS LORD Microstrain IMU connected to a Rapsberry Pi 3B+ unfortunately without success.

I compiled the MSCL_Inertial_Example_C++ program (that I renamed MSCL) modifying the serial port to /dev/ttyUSB0 and the baud rate to 115200 by using the following instruction:
   
mscl::Connection connection = mscl::Connection::Serial("/dev/ttyUSB0",115200);
   
When I run the example program by using:

export LD_LIBRARY_PATH=/usr/share/c++-mscl:/usr/share/c++-mscl/Boost/lib:$LD_LIBRARY_PATH; ./MSCL

it prints:
   
Node Information: 
Segmentation fault

For your reference I'm using:
- MSCL library: c++-mscl_65.0.0_armhf.deb
- Raspberry Pi OS(Legacy) Lite

If I give the command "uname -m" I obtain the reply: armv7l

If I use gdb I obtain the following output:

root@raspberrypi/home/pi/src/MSCL# export LD_LIBRARY_PATH=/usr/share/c++-mscl:/usr/share/c++-mscl/Boost/lib:$LD_LIBRARY_PATH;gdb ./MSCL

GNU gdb (Raspbian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./MSCL...
(No debugging symbols found in ./MSCL)
(gdb) run
Starting program: /home/pi/src/MSCL/MSCL 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x76740440 (LWP 1408)]
[Thread 0x76740440 (LWP 1408) exited]
Node Information: 

Thread 1 "MSCL" received signal SIGSEGV, Segmentation fault.
0x76d794e4 in mscl::Connection::byteAppendPos() const () from /usr/share/c++-mscl/libmscl.so
(gdb) bt
#0  0x76d794e4 in mscl::Connection::byteAppendPos() const () from /usr/share/c++-mscl/libmscl.so
#1  0x76d7e6fe in mscl::ResponseCollector::registerResponse(mscl::ResponsePattern*) () from /usr/share/c++-mscl/libmscl.so
#2  0x76d7e1da in mscl::ResponsePattern::ResponsePattern(std::weak_ptr<mscl::ResponseCollector>) ()
   from /usr/share/c++-mscl/libmscl.so
#3  0x76dd8f5c in mscl::GenericMipCommand::Response::Response(mscl::MipTypes::Command const&, std::weak_ptr<mscl::ResponseCollector>, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned char) ()
   from /usr/share/c++-mscl/libmscl.so
#4  0x76dd75a6 in mscl::GetDeviceInfo::Response::Response(std::weak_ptr<mscl::ResponseCollector>) ()
   from /usr/share/c++-mscl/libmscl.so
#5  0x76dbd4c2 in mscl::MipNode_Impl::getDeviceInfo() const () from /usr/share/c++-mscl/libmscl.so
#6  0x76dcb6a8 in std::_Function_handler<mscl::MipDeviceInfo (), std::_Bind<std::_Mem_fn<mscl::MipDeviceInfo (mscl::MipNode_Impl::*)() const> (mscl::MipNode_Impl const*)> >::_M_invoke(std::_Any_data const&) () from /usr/share/c++-mscl/libmscl.so
#7  0x76dc9434 in mscl::MipNodeInfo::deviceInfo() const () from /usr/share/c++-mscl/libmscl.so
#8  0x76db8e36 in mscl::MipNode_Impl::modelName[abi:cxx11]() const () from /usr/share/c++-mscl/libmscl.so
#9  0x76dd0750 in mscl::MipNode::modelName[abi:cxx11]() const () from /usr/share/c++-mscl/libmscl.so
#10 0x00013acc in main ()
(gdb) quit

Do you have any idea where the problem can be?

Thank you.

msclissa, 2023-11-27:
Sorry you're running into this issue! Unfortunately, I'm really not sure of the cause. I have not tested on your specific setup, but we have not run into this problem on other Linux environments. It's probably not the issue, but can you double check that you have both read and write permissions to the port the device is connected over? I guess it's also possible the program is running into resource issues and cannot allocate the memory it needs.

I'll do some testing and hopefully find some answers.

Another option you might consider is switching over to our new [MIP SDK](https://github.com/LORD-MicroStrain/mip_sdk/tree/develop) | [Build Instructions](https://github.com/LORD-MicroStrain/mip_sdk/tree/develop#how-to-build). This is a new API in development specifically for our inertial product line - at this point both the MIP SDK and MSCL are being maintained alongside one another but eventually we will be shifting support and focus to the MIP SDK. The MIP SDK is being developed with the goals of being light-weight (no Boost!) and inertial-focused. If you'd like to try to switch over we'd be happy to help get you up and running! MIP SDK example code for CV7 can be found [here](https://github.com/LORD-MicroStrain/mip_sdk/tree/develop/examples/CV7).

Please note that the develop branch of the MIP SDK project should be used at this point to ensure access to the latest device features and updated examples - we are ironing out our release process so things have not been moved to main in a while.

gabriele-bruzzone, 2023-11-28:
Thank you very much for your support.
I triple checked the wiring of the IMU and the write and read permissions of the serial devices on my Raspberry Pi 3B+ system but unfortunately the problem with the MSCL persists.
I also tested the IMU by using the same test program and the same version of the MSCL library that I installed on the Raspberry (of course the x64 version instead of the armhf version) on an laptop running Ubuntu 22.04 and it worked.
So still no joy with the MSCL on the Raspberry Pi.
Then, as you suggested, I installed the new MIP SDK on the Raspberry Pi and the example programs (both the CV7_example.c and the CV7_example.cpp) worked like a charm with the IMU.
I will need to do a further effort to convert the programs I already wrote using MSCL API calls to use the MIP SDK API calls, anyway this is a possible workaround to solve my problem.
In any case, I would be very curious to discover what causes the problem of the MSCL with the Raspberry Pi.

msclissa, 2023-11-28:
I'm glad to hear the MIP SDK is working for you on the Raspberry Pi!
We have gone through the process of switching a couple of our internal tools over from MSCL to the MIP SDK and it has been pretty straightforward for the most part but definitely let us know if you run into any confusion!

I did not end up having time today to look into tracking down the armhf/Raspberry Pi seg fault issue but should be able to look into it before the end of the week.

closed 2024-10-24 by github-actions[bot]