LORD-MicroStrain/MSCL #371: Failed to read the Model Number (EEPROM 46) from Node

sgacond, 2023-11-22:
Hello!

I have a problem communicating with an SG-LINK-200 via MSCL (Python).
I followed the sample and use the initialization sequence like in this sample: [Configure Node (Python)](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL_Examples/Wireless/Python/ConfigureNode.py)

- `setToIdle`
- wait until `mscl.SetToIdleStatus.setToIdleResult_success` 
(works until here)

.. and then configuring the nodes like:
```python
        config.defaultMode(mscl.WirelessTypes.defaultMode_idle)
        config.inactivityTimeout(7200)
        config.samplingMode(mscl.WirelessTypes.samplingMode_sync)
        config.dataCollectionMethod(3)
        config.sampleRate(118)
        config.unlimitedDuration(True)
```

Unfortunately, applying the configuration always fails with: `Failed to read the Model Number (EEPROM 46) from Node 36170` .
I'm not exactly sure if its the verify or apply call.... But one the two.

The node was in transmission mode before and transmitted data sucessfully. The error appeared doing a restart of the software using the MSCL lib. Therefore, I think, wireless issues shouldn't be the cause here.

The error persists. I can do multiple restarts, always the same.
Am i doing something wrong during the initialization sequence? Is there another place where more delays or waiting should be done?

msclissa, 2023-12-04:
It sounds like your initialization sequence is correct. Just to clarify - the `SetToIdleStatus` does transition to `setToIdleResult_success` and the LED on the node goes to a pulse pattern to signify it is in idle mode?

Based on the fact it was transmitting data and the base station was receiving that data successfully before I agree it doesn't seem like an issue with the frequency configuration or interference in wireless communication.

Do you have other nodes to test with - is this initialization sequence successful on another node?

Also, if you have access to a Windows machine that can run [SensorConnect](https://www.microstrain.com/software/sensorconnect) that may be helpful in seeing the state of the node and troubleshooting if there is something going wrong with the node itself. If possible, can you try to connect to the node through SensorConnect to confirm the node is idle and try to change a couple settings to verify communication is successful?

If these troubleshooting steps don't shed any light on to what's going wrong it may be helpful if you sent your exact initialization code so I can try to recreate the issue here.

Thanks!

sgacond, 2023-12-05:
Hi @msclissa 

Thanks for your response!

Yes, `setToIdleResult_success` was returned (also within a normal time span).

We did some tests with another node, another USB dongle and also with SensorConnect. 

Unfortunately, the results were not very consistent:
- The mentioned issue was always with the same node
- SensorConnect usually worked correctly. Little issues also happened sometimes, but no persistent, unrecoverable error.
- This persistent error after setToIdle (which was also not recoverable without resetting the node) was only present when using the SDK. It wasn't always reproducable. But when it's there, it stays.

I was not sure about the correctness of my initialization sequence. Is there more documentation about it, then these [examples here](https://github.com/LORD-MicroStrain/MSCL/tree/master/MSCL_Examples/Wireless/Python)?

closed 2024-10-24 by github-actions[bot]