LORD-MicroStrain/MSCL #380: Raspberry pi5 | ImportError: /_mscl.so: ELF load command address/offset not page-aligned

JosHurt, 2024-02-22:
Good afternoon, I am currently working on a project that involves the integration of a Raspberry Pi 5 with a 64-bit operating system, with the purpose of connecting a WSDA-200-USB gateway. During development, I am facing difficulties while trying to use the MSCL API. There is a problem when importing the mscl library specifically on an arm64 architecture and using Python 3.

The specific error I'm encountering manifests itself as follows:

ImportError: /_mscl.so: ELF load command address/offset not page-aligned

I have made several attempts to address this situation, including modifying between Python versions 2.7.18, 3.6.15 and 3.11, but unfortunately the same error persists in both instances.

I appreciate any guidance or assistance you can provide to resolve this issue. Thank you very much for your attention and support.

jmontp, 2024-04-04:
I am also facing the same issue with the 64 bit version of Raspbian. It seems to be working fine for 32bit installations though. 

Any updates would be appreciated.

jmontp, 2024-04-04:
I found a temporary fix to this problem by adding 

`kernel=kernel8.img`

in /boot/firmware/config.txt. This reverts the page size back to 4k from 16k. [Link to a better explanation of the issue.](https://github.com/raspberrypi/bookworm-feedback/issues/120)

closed 2024-10-24 by github-actions[bot]