LORD-MicroStrain/MSCL #387: Using Communication protocol command sets

jagan-pm, 2024-04-17:
By using the 3DX-GX5-AHRS communication protocol commands as referenced in the manual, I created a base code. However, it's not working. Could you please suggest a solution?
[sample_code.txt](https://github.com/LORD-MicroStrain/MSCL/files/15007345/sample_code.txt)

msclissa, 2024-04-29:
Apologies for the delay!

Based on your email request you attempted to use MSCL but it didn't work for you. Please provide more information about what went wrong - were you able to import MSCL but then errored when trying to connect to or communicate with the sensor? Where there runtime errors? Did everything run successfully but did not result in the expected behavior?

The code you provided does not utilize MSCL or any of our other APIs. Developing your own parser is certainly possible but we can only offer limited support (clarifying protocol structure, assistance determining the right commands to use) - I would recommend using MSCL for communication via Python.

You can find pre-built MSCL Python packages for various versions and environments here: [Pre-built MSCL Binaries/Packages](https://github.com/LORD-MicroStrain/MSCL?tab=readme-ov-file#pre-built-mscl-binariespackages)

Our setup and installation guide can be found here: [How to use MSCL](https://github.com/LORD-MicroStrain/MSCL/blob/master/HowToUseMSCL.md)

Python examples for inertial devices to help get you started can be found here: [MSCL_Examples/Inertial/Python](https://github.com/LORD-MicroStrain/MSCL/tree/master/MSCL_Examples/Inertial/Python)
The setCurrentConfig.py example includes how to configure output data values and rates. Keep in mind that different devices support different commands and data values - if you see errors indicating things are unknown or not supported those can be easily resolved by looking in the communication protocol documentation you've been using to see what is supported on the device or reaching out to us - sometimes the names of things aren't exactly the same in the protocol documentation and MSCL, we would be happy to help figure out what functions or data values you're looking for!

If you run into any issues it would be helpful to know:
- The environment you are running in (operating system, Python version)
- What behavior you are trying to implement
- What exactly is failing and any associated error messages
- Anything you have tried to resolve the issue

Hopefully this helps clarify things! Let us know if you continue to run into any issues!

jagan-pm, 2024-04-30:
No problem at all...! It sounds like you're trying to set the sample rate using the example provided in setCurrentConfig.py, but it's not executing properly and showing the error "command (3399) is not supported" in Python.

jagan-pm, 2024-05-06:
No problem at all...! It sounds like you're trying to set the sample rate
using the example provided in setCurrentConfig.py, but it's not executing
properly and showing the error "command (3399) is not supported" in
Python.suppose i want only 100 samples per seconds .and this message i was
put in git as a comments please check.

thank you
jagadees p m

closed 2024-10-24 by github-actions[bot]