LORD-MicroStrain/MSCL #390: Timestamp issue

8c1h9q7hwz0cl, 2024-05-07:
Hello, we are facing some inconsistent timestamp issues. We configured the IMU to run at 200Hz. In the ideal situation, the time difference (dt) between two consecutive IMU measurements should be 5ms. However, we observed that the dt varies between 3 and 7 ms. 
[image]

gmurrayrv9a, 2024-05-08:
We are using MSCL and getting the timestamp as follows:
       for(mscl::MipDataPacket packet : packets)
        {
           //get the data in the packet
            mscl::MipDataPoints data = packet.data();

            mscl::MipDataPoint dataPoint;
            mscl::Timestamp packet_ts = packet.collectedTimestamp(); 
   
Then we use “packet_ts.nanoseconds()”

Looking at MSCL code we see that time stamps are not associated with the received accelerometer or gyroscope packets but rather stand alone in a 'internal timestamp' packet. There is code (in MSCL) to calculate delta from previous timestamp, add this to PC time and then attach this value to accelerometer/gyroscope data as it comes in. Sometimes multiple accelerometer packets are received before a time stamp packet causing multiple accelerometer data to have the same timestamp.

gmurrayrv9a, 2024-05-09:
We have found a resolution that works for us. We enabled CLASS_SYSTEM channel 0xA0D5 to get TICKS and calculate our own timestamp using system time and TICKS.

closed 2024-10-18 by dacuster