LORD-MicroStrain/MSCL #391: How to disable the magnetometer for GV7 models?

karlji1021, 2024-05-11:
Hello, I recently changed the IMU model from 3DM-GX5-AHRS to 3DM-GV7-AHRS.
Before changing the model, I used a code snippet like the one below to disable a magnetometer.

"""
mscl::HeadingUpdateOptions headingUpdateOptions;
headingUpdateOptions.useInternalMagnetometer = false;
node_->setHeadingUpdateControl(headingUpdateOptions);
"""

However, for the GV7 model, "setHeadingUpdateControl" method doesn't work anymore.
The method gives "Error_NotSupported" error.

So, what I've tried to solve this problem are the ones below.

1. Disable aiding measurement source
"""
enableDisableAidingMeasurement(InertialTypes::AidingMeasurementSource::MAGNETOMETER_AIDING, false);
"""

2. Disable magnetometer usage in SensorConnect.
The magnetometer usage was set to be disabled, and this setup is saved as the initial settings.
I've checked that the disabled setting remained after disconnecting/reconnecting the IMU cable.

To test if the magnetometer is disabled, a small magnet was put close to the IMU.
Then, the estimated direction changed depending on the direction of the magnet.

I wonder how to completely disable its magnetometer.

Thank you.

msclissa, 2024-06-03:
Apologies for the delayed response!

As far as I'm aware, the magnetometer should be disabled as an aiding source after setting the Aiding Measurement Control (0x0D,0x50) configuration to false for the magnetometer aiding source (0x0004), which you have correctly implemented:
```
enableDisableAidingMeasurement(InertialTypes::AidingMeasurementSource::MAGNETOMETER_AIDING, false);
```
Especially since you've confirmed that the setting persists through power cycle.

It sounds like  you are still seeing unexpected data output/behavior despite this configuration. I'm going to direct you to our general support portal [https://support.microstrain.com/](https://support.microstrain.com/) where our inertial product experts can help you figure out why the device is behaving in this way and what additional settings may be needed to achieve the expected results! I promise their response times are much better than what you have found here!
__________________

I know you already have this working, just want to include it here for reference in case it's helpful:
**Save settings to persist through power cycle via MSCL**
```
// save just the aiding measurements (all sources) enable/disable state as startup
device.saveSettingsAsStartup({mscl::MipTypes::CMD_EF_AIDING_MEASUREMENT_ENABLE});

// save all current device settings as startup
device.saveSettingsAsStartup();
```

karlji1021, 2024-06-04:
Thank you for the kind reply!

Yesterday, I found out that I also had to disable "north compemsation".
In SensorConnect, I disabled "enable north compensation" and finally get the result which I wanted.
The imu is not affected by external manets anymore.
Does this function uses other source than manetic fields?
And, is there any function to disable the north compensation in mscl library?

Best regards,
Gwanghyeon Ji

closed 2024-10-24 by github-actions[bot]