LORD-MicroStrain/MSCL #392: CPU utilisation very high

jagan-pm, 2024-06-26:
I am running Python code to extract Roll, Pitch, and Yaw from a Microstrain 3DM-GX5-AHRS and store the data with timestamps. However, running this code consumes more than 90% of the CPU. How can I optimize and reduce the CPU utilization?
my system details.
Operating System: Ubuntu Mint 20.04
Software: Python 3
Hardware: Intel Atom Dual Core Processor SBC
RAM: 8 GB
Storage: 64 GB SSD
[optimised_v2.txt](https://github.com/user-attachments/files/15987395/optimised_v2.txt)

msclissa, 2024-07-01:
Have you done any profiling to figure out what is causing the utilization to run high? Would you mind replacing `node.getDataPackets()` and the subsequent loop with
```
# look at how many packets node.getDataPackets() usually returns, probably averaged over a couple iterations
EXPECTED_PACKETS = 1

for i in range(EXPECTED_PACKETS):
    data_points.append(["0.0", "0.0", "0.0"])
```
Also, remove
```
node = DisplacementNode(connection)
```
to prevent continuous data collection and buffering.

Is CPU utilization still high when MSCL is removed from the equation?
Leaving the CSV open and just writing to it every loop instead of re-opening it every loop might also help?

A quick aside, I would recommend using `InertialNode` instead of `DisplacementNode` - it won't make a difference in this scenario because they use the same parser and `getDataPackets()` function, but `InertialNode` will have all the command interface functions your device supports.

jagan-pm, 2024-07-24:
Dear  msclissa
Please offer a proper solution for this .
I contacted your team regarding very high CPU utilization, but I have not
received a proper solution. I also posted the issue in the Git repository
along with the program.

Currently, I am trying to acquire data using the DCP manual for the
3DM-GX5-25 AHRS sensor. Could you please provide the commands or a sample
program to acquire roll, pitch, and yaw data from the sensor?

Thank you.

closed 2024-10-24 by github-actions[bot]