LORD-MicroStrain/MSCL #393: MCSL TIming

taylorpool, 2024-06-26:
Hi all,

The situation is the following:

The MSCL API for a MipDataPacket has two member functions:

`deviceTimestamp()` gives the timestamp associated with the most recent time given by the IMU. On our system, this only reports to the millisecond range.
`collectedTimestmap()` gives the timestamp associated with when the host computer received the message from the IMU (not when the `MipPacket` was measured). This timestamp reports to the nanosecond range.

We need precise timing beyond the millisecond range for each gyro/acceleration measurement received from the IMU. How can we achieve this?
I have attached an image of my terminal output printing each IMU measurement. You can see the collected timestamp, the different between the collected and device timestamps (due to delays in serial communication), and the timestamp associated with device timestamp. All timestamps are in nanoseconds.

Please advise how to proceed.

[image]

taylorpool, 2024-10-03:
We are still suffering from this issue, and any feedback/advice would be greatly appreciated.

dacuster, 2024-10-17:
Hi @taylorpool,

Both of those functions returns a [Timestamp](https://lord-microstrain.github.io/MSCL/Documentation/MSCL%20Documentation/index.html#File:Timestamp.h:Timestamp) object which has functions to get the time in any format down to [nanoseconds](https://lord-microstrain.github.io/MSCL/Documentation/MSCL%20Documentation/index.html#File:Timestamp.h:Timestamp.nanoseconds). I see you are getting the [difference](https://lord-microstrain.github.io/MSCL/Documentation/MSCL%20Documentation/index.html#File:Timestamp.h:Timestamp.-) in those times which returns a [Timespan](https://lord-microstrain.github.io/MSCL/Documentation/MSCL%20Documentation/index.html#File:TimeSpan.h:TimeSpan) object which also has functions to get the format down to [nanoseconds](https://lord-microstrain.github.io/MSCL/Documentation/MSCL%20Documentation/index.html#File:TimeSpan.h:TimeSpan.getNanoseconds).

Hopefully that's enough information to get you what you need. If you have further questions a code snippet would help in diagnosing your problem.

closed 2024-10-24 by github-actions[bot]